dukeduck1984/uReflowOven-Esp32-Micropython #8: 搜索不到WIFI信号

BG7YWL, 2020-07-06:
使用ESP32烧录座+ESP32-WROVER-B模块的方式烧录固件发现无法搜索到WIFI型号，ES32-WROOM-32模块也有尝试
[image]
[image]
[image]

dukeduck1984, 2020-07-06:
我试了手头的两块Esp32 Wroom，都没有问题呀
```
>>> import network
>>> sta = network.WLAN(network.STA_IF)
>>> sta.active(True)
I (68968) phy: phy_version: 4102, 2fa7a43, Jul 15 2019, 13:06:06, 0, 0
True
>>> sta.scan()
[(b'28#301_ASUS', b'p\x8b\xcd\xc8\xba ', 1, -20, 3, False), (b'28#301_ASUS', b'\x0c\x12bb\xb7{', 1, -47, 3, False), (b'28#301_MI', b'\xecA\x18]
>>>
```

自己作为AP，或者作为station去连接别的AP也都正常

你贴的图上我也没看到报错信息

BG7YWL, 2020-07-07:
通过AMPY工具，我已经将MAIN目录下所有文件、文件夹都上传到ESP32中，但是还是没有WIFI信号。
[image]

另外我参考config.json文件，只看出以下的接线，有部分引脚接线不知道怎么接，比如TFT_ACC和MAX31855模块的HWSPI、CS、MISO、SCK，我按照你说的模块购买上面并没有这些引脚。
| TFT | ESP32 |
| ------------ | ------------ |
| T_IRQ | 35 |
| T_DO |   |
| T_DIN |   |
| T_CS | 25 |
| T_CLK |   |
| MISO | 5 |
| LED | 15 |
| MOSI | 18 |
| DC | 12 |
| RST | 4 |
| CS | 13 |
| GND | GND |
| VCC | 3V3 |

| BUZZ | ESP32 |
| ------------ | ------------ |
| GND | GND |
| I/O | 22 |
| VCC | 3V3 |

| MAX31855 | ESP32 |
| ------------ | ------------ |
| GND | GND |
| VCC | 3V3 |
| DO |  |
| CS |  |
| CLK |  |

dukeduck1984, 2020-07-07:
TFT的ACC是给屏幕供电的GPIO，接到屏幕的VCC上，但实际上你把屏幕的VCC直接接到3.3V即可，程序本身并没有关闭屏幕的操作。T_DO直接并联到TFT的MISO，T_DIN并联到TFT的MOSI，T_CLK并联到TFT的SCK。

MAX31855的DO就是MISO，CLK就是SCK。hwspi是硬件spi的编号，不是GPIO，mpy文档里有写。

你硬件线没接好的话，main.py执行的时候肯定在初始化硬件的时候就出错终止了，不会执行到最后打开AP信号的代码，所以没有wifi信号。

closed 2020-07-08 by BG7YWL