dukeduck1984/uReflowOven-Esp32-Micropython #9: Stages messed up

pastaclub, 2020-07-08:
I tested the new code (with your PID adjustments). It's not working well on my hardware.

The actual temperature lacks behind the profile and it switches stages much too late. On my first run, it was still in pre-heat stage when it should already have reached the reflow peak, then it froze and didn't react anymore (I had to turn off the power and restart). On the second test run, it stayed in pre-heat for too long and the temperature actually dropped again. Much later it started heating again and reached soak only when reflow was supposed to be over.

Obviously the PIDs need to be changed, but I am wondering whether it is correct that the machine doesn't change the stages when the time is reached. It seems that it's waiting for the specified temperature. Maybe it should switch stages when the time of the stage has elapsed?

[image]

dukeduck1984, 2020-07-09:
Yes, the stage status changes when the it reaches certain temp which is defined by the "stages" in the profile json file.  What it does is just letting user know the stage status of the process rather than making a real impact on temp control, except that the integration part of the PID is only enabled when it reaches 'reflow' stage.

The target temp the temp control logic tries to sync with is defined in "profiles" in profile json, along with the corresponding timeline.

That was implemented by adafruit in the original code, and I don't think it's not reasonable, nor can I think of a better solution, so I didn't change it.

I assume tuning the PID params should improve the performance of your hardware.  I would suggest increase Kp and Ki, and decrease Kd.

pastaclub, 2020-07-11:
As it turned out, the problem with the curve falling could be fixed by PID tuning. Yet, I don't find it logical that the displayed stage changes with the reaching of the temperature. The stage is specified by the profile, so it would be more logical to change the state once the time is reached. If the temperature doesn't reach the specified value because the PIDs are wrong (as in the photo above) that is a different problem to fix, but still, I would expect the machine to report the phase specified in the profile.

dukeduck1984, 2020-07-12:
I'm not sure which way is more logical but think it not a big problem as long as the actual temp can be well controlled to meet the ideal profile.