dukeduck1984/uReflowOven-Esp32-Micropython #11: 运行遇到一点小报错

BG7YWL, 2020-08-17:
暂时还没有接SSR，温度检测显示X_GND，拔掉K型热电偶显示NC，另外蜂鸣器会有微弱的“哒哒哒哒”声响。

这是我焊接的一块板子
[image]

以下是我的接线方式
| TFT-3.2 | ESP-32 |
| --- | --- |
| T_IRO | 35 |
| T_DO | 5 |
| T_DIN | 18 |
| T_CS | 25 |
| T_CLK | 19 |
| MISO | 5 |
| LED | 15 |
| SCK | 19 |
| MOSI | 18 |
| DC | 12 |
| RST | 4 |
| CS | 13 |
| GND | GND |
| VCC | 14 |

| BUZZ | ESP-32 |
| --- | --- |
| GND | GND |
| I/O | 22 |
| VCC | VCC |

| MAX31855 | ESP-32 |
| --- | --- |
| GND | GND |
| VCC | VCC |
| DO | 33 |
| CS | 27 |
| CLK | 32 |

| SSR | ESP-32 | AC220V |
| --- | --- | --- |
| 1 | * | 负极 |
| 2 | * | 正极 |
| 3 | GND | * |
| 4 | 21 | * |

dukeduck1984, 2020-08-18:
先确认下你的MAX31855接线没有问题。

启动过程中显示“Emergency Off”与MAX31855有关，见`oven_control.py`的第63行的`get_temp()`。

你可以尝试把`max31855.py`里第54-59行注释掉：
```
        # if self._data[3] & 0x02:
        #     raise RuntimeError("X_GND") # shortcut to GND
        # if self._data[3] & 0x04:
        #     raise RuntimeError("X_PWR") # shortcut to power
        # if self._data[1] & 0x01:
        #     raise RuntimeError("ERR") # faulty reading
```

注释后看看能不能正常显示温度，如果还是不能，那应该是模块或者探头的问题。

dukeduck1984, 2020-08-18:
蜂鸣器我用的跟你一样，我的蜂鸣器在屏幕显示启动时会发出“哒”的一声，之后似乎没有什么噪音。

BG7YWL, 2020-08-19:
注释掉代码，温度能显示，不过显示的不是正确温度，一直在乱跳。换了另一款MAX31855模块温度就显示正常了。
蜂鸣器我换了一个新的，哒哒哒的声音没有那么响了，应该是我把蜂鸣器的90度排针换成直插排针的时候把蜂鸣器弄坏了。

dukeduck1984, 2020-08-19:
你这温度曲线看上去不错嘛:-)

没问题我关issue了，有其他问题你再开

closed 2020-08-19 by dukeduck1984

BG7YWL, 2020-08-20:
我用的是康佳（KONKA）KAO-1208烤箱，京东买的只需99元。