dukeduck1984/uReflowOven-Esp32-Micropython #12: uReflowOven PCB

Bibbbi, 2020-08-25:
### Here is a suitable board for the uReflowOven project

- LILYGO TTGO T8 (8MB PSRAM / 4MB FLASH / TF-CARD) [https://www.aliexpress.com/item/32851423415.html](url)
- 2 x MAX31855 Thermocouple-to-Digital Converter
- 4 x digital outputs for HEATER, BUZZER, FAN and BOOSTER
- 3 x digital inputs for operation with a non-touch screen panel
- It is possible to mount an ILI9341 2.8 "or a 3.2" display [https://www.aliexpress.com/item/32960934541.html](url)
- +5Volt Power Supply Jack DC005 5.5x2.1mm barrel type [https://www.aliexpress.com/item/4000198429452.html](url)

Download Gerber file: [uReflowOven_Gerber_2020-08-25.zip](https://github.com/dukeduck1984/uReflowOven-Esp32-Micropython/files/5126076/uReflowOven_Gerber_2020-08-25.zip)

[image]
[image]
[image]
[image]
[image]

### Pictures of my prototype:

[image]
[image]
[image]

dukeduck1984, 2020-08-29:
Nice work!  Thanks for sharing!

funfly28, 2020-10-02:
Can provide BOM？Thanks

duojie, 2020-11-04:
Can provide BOM？Thanks

Bibbbi, 2021-01-30:
@pastaclub The touch pins are connected. The hardware buttons are optional. If you use a display without touch function.

**BOM List:**
|   	|   Value	|  Case 	|
|---	|---	|---	|
|  C1, C2, C3, C4, C6 | 100nF | SMD-0805 | 
|  C5	|   47uF 25V |   SMD-D 7343 | 
|  D1, D2, D3, D4, D5 | MBR120VLSFT1G | SOD-123FL	| 
|  J1 | DC-005 | DC Power Jack | 
|  R1, R2, R3, R4 | 1K2 Resistor | SMD-0805 | 
|  R5, R6, R7 | 10K Resistor | SMD-0805 (optional) | 
|  T1, T2, T3, T4 | MMBT2222A NPN | SOT-23 | 
|  X1 | KF301-4P | Terminal Red 5,08mm | 
|  X2 | KF301-8P | Terminal Blue 5,08mm |

pastaclub, 2021-02-01:
Thank you! I ordered the transistors and terminals (I had compatible components for the rest). I will get the PCB made and see how it works for me. Maybe that also fixes my problem with the sensor.

We could add the schematics and board files to the repository if you want to share them.

Zefram88, 2021-02-17:
According to the schematics and PCB, IRQ and CS pins on touchscreen are not connected anywhere, how can it work?