dukeduck1984/uReflowOven-Esp32-Micropython #13: Plans for next version

dukeduck1984, 2021-01-27:
Just a few thoughts, no concrete timeline now.  No guarantee, it may not happen at all.

* Upgrade to LVGL v7 (or v8 if released) and the lastest stable MicroPython
* For scheduler mechanism, replace `Timer` & `_thread` with `uasyncio`.
* Adding servo & fan control options to open the door of the oven and speed up cooling.
* Plug & play, no need to modify the internal of the oven.

Bibbbi, 2021-01-27:
Great, I can't wait for that to happen.

nmcq, 2021-01-30:
继电器是用交流控制交流的吗？？？

dukeduck1984, 2021-01-30:
> 继电器是用交流控制交流的吗？？？

直流控交流，需要能以直流3V触发，否则就得加mos或者三极管了

nmcq, 2021-01-30:
> > 继电器是用交流控制交流的吗？？？
> 
> 直流控交流，需要能以直流3V触发，否则就得加mos或者三极管了

感谢您的回复。

nmcq, 2021-02-05:
Plug & play, no need to modify the internal of the oven.
目前版本将烤箱开到最大温度，长通状态（机械烤箱），直接控制烤箱电源应该也不用打开烤箱修改内部吧？

dukeduck1984, 2021-02-05:
> Plug & play, no need to modify the internal of the oven.
> 目前版本将烤箱开到最大温度，长通状态（机械烤箱），直接控制烤箱电源应该也不用打开烤箱修改内部吧？

对，你把所有电子器件装在一个盒子里，安一个国标三眼插座，把烤箱电源线接上去即可。

Zefram88, 2021-02-21:
I find also pretty uncomfortable using the AP to access FTP, should be possible to add the ability to connect to an existing network? I tried add it by myself but code crashes somewhere, probably because of IP binding.

dukeduck1984, 2021-02-21:
Well, the reason to use AP mode is because it's most straight forward and needs very little code, and doesn't have to deal with handling wrong SSID or password etc., after all it's meant for uploading code / debugging purpose only - you don't need a wifi to do the soldering work.

dariomicro, 2021-12-01:
> Just a few thoughts, no concrete timeline now. No guarantee, it may not happen at all.
> 
> * Upgrade to LVGL v7 (or v8 if released) and the lastest stable MicroPython
> * For scheduler mechanism, replace `Timer` & `_thread` with `uasyncio`.
> * Adding servo & fan control options to open the door of the oven and speed up cooling.
> * Plug & play, no need to modify the internal of the oven.
> * It might worth adding an ESP8266 for the sake of safety to monitor the temperature in case the ESP32 crashes, the power can be cut.

hello, your work is great, I have been programming microcontrollers for a long time but I have no knowledge of linux and it is very difficult for me to compile LVGL + micropython.
Could you do this job for me?
I will reward you for this ($)

Thank you..
microlux.dario@gmail.com

dukeduck1984, 2021-12-02:
> > Just a few thoughts, no concrete timeline now. No guarantee, it may not happen at all.
> > 
> > * Upgrade to LVGL v7 (or v8 if released) and the lastest stable MicroPython
> > * For scheduler mechanism, replace `Timer` & `_thread` with `uasyncio`.
> > * Adding servo & fan control options to open the door of the oven and speed up cooling.
> > * Plug & play, no need to modify the internal of the oven.
> > * It might worth adding an ESP8266 for the sake of safety to monitor the temperature in case the ESP32 crashes, the power can be cut.
> 
> hello, your work is great, I have been programming microcontrollers for a long time but I have no knowledge of linux and it is very difficult for me to compile LVGL + micropython. Could you do this job for me? I will reward you for this ($)
> 
> Thank you.. [microlux.dario@gmail.com](mailto:microlux.dario@gmail.com)

[Here](https://github.com/dukeduck1984/uReflowOven-Esp32-Micropython/tree/master/FIRMWARE) you will find the builds of the firmware which will work with the current stable release.

NOTE: I have been very busy with my own job, so pls consider this repo not under active development, the code will not work with the most current MicroPython and LVGL.

sle118, 2023-01-24:
Just wanted to add my two cents here for anyone interested. I have done a great deal of retrofitting to the newest lv_micropython build in order to be able to use it on a marker fabs ESP32 3.5" capacitive touch screen (https://github.com/Makerfabs/Project_Touch-Screen-Camera). Under the new framework, timers were causing a great deal of instability, so I moved the oven control as well as some idle temperature update to threads which use the same cadence mechanism (e.g. rate configured in JSON file) and now everything is rock solid.

Some other notable changes are, for example, using a grid for the main layout, leveraging the chart capabilities to draw the profile, melting temperature as well as vertical cursors crossing the profile line at each critical steps. The capacitive panel doesn't require calibration, so although the settings menu that allows calibration or PID configuration selection was preserved, but hitting the settings button from a capacitive screen will skip it and show the PID and offset entry. That one uses the "window" object at its core, and keeps the numerical keyboard as is.

Overall, the look and feel has shifted away from the original (I'm using the base theme from lvgl), but all the controls are where they were placed originally, as their positioning made a lot of sense.

sle118, 2023-01-25:
Well, I've done my best with micropython has waaay too much overhead for the sake of my sanity. So I've decided to see if I can port this to a C project. In order to make it more "accessible" to the general public (Python does seem easier to some extent), I'll make sure to use the arduino framework, which has a ton of useful libraries. Stay tunes for the uReflowOven-ESP32-C

sle118, 2023-01-26:
I almost did a rage quit on mycropython because the system kept dumping on my with a stack overflow on the mp_task. I ended up digging quite a bit only to figure out what was going on; it seems like the mp_task scheduler was getting initialized at the very end of the main.py script when I was starting the buzzer thread.... with a relatively small stack size given its simplicity. So I am back into fine tuning navigation, etc with no more crashes this time. I'll also look at the beta 2 version to see if I need to retrofit anything else more recent and finally I'll add an option to control a GPIO that will stay on during the whole cycle (in my case, it's to drive a convection fan).

If there is any interest here, I'll go ahead and create a pull request when I'm satisfied.

pastaclub, 2023-01-27:
Sure, let us see it! MicroPython can be a pain sometimes... it's mainly easier to interactively develop with it, but we also ran into memory problems in the past. A C implementation is definitely interesting.

sle118, 2023-01-29:
I will definitely not have to go down the route of C as thorough debugging and tracing led me to figuring out how memory was being allocated and now the micropython version (ported to the latest lv_micropython) is rock solid. I took a long détour porting this project over because I liked the look and feel and it's features, but the point was to have an oven that would allow me to finish my current project, a multi room music steamer with Ethernet, optical and RCA digital out, analog out and analog in running the firmware I co authored, squeezelite-esp32