dukeduck1984/uReflowOven-Esp32-Micropython #15: wifi有信号但无法链接

nmcq, 2021-02-04:
![Uploading 1612455086894.jpg…]
另麦克风有哒哒哒哒声。温度用体温测正常。我用的6675。
目前还有继电器未链接。
[image]

dukeduck1984, 2021-02-05:
wifi你试下别的电脑，或者用手机试下是否能连接。从来没遇到过wifi连不上的问题。你用的是什么模块？

蜂鸣器应该是是无源的，你用的蜂鸣器什么样？拍个照看看

nmcq, 2021-02-05:
[image]
[image]
[image]

nmcq, 2021-02-05:
以上是蜂鸣器图片，手机也无法链接wifi，我用的是，T8 V1.8 ESP32-WROVER 8MB PSRAM。另外热电偶遇到低温时似乎响应比较慢，插入冷水中，要等个一分钟左右，温度才会降下去。

nmcq, 2021-02-05:
[image]
手机提示，网络拒绝加入

dukeduck1984, 2021-02-06:
* Wifi的问题真的很奇怪，我从来没遇到过类似情况。我建议你去[官方论坛](https://forum.micropython.org)发帖求助。因为无线热点是直接使用micropython自己的`network`模块创建的，并无特殊之处，代码可以在`main.py`最下面找到。另外我建议你先不运行任何程序，串口连上esp32，进入REPL，在命令行创建一个无线AP看看是否能连上。

* 蜂鸣器硬件看着跟我用的是一样的，供电是几伏？电源从哪里接的？

* 热电偶我感觉多数是硬件本身问题了，你尝试换个模块或者探头看看是否有区别。

nmcq, 2021-02-06:
> * Wifi的问题真的很奇怪，我从来没遇到过类似情况。我建议你去[官方论坛](https://forum.micropython.org)发帖求助。因为无线热点是直接使用micropython自己的`network`模块创建的，并无特殊之处，代码可以在`main.py`最下面找到。另外我建议你先不运行任何程序，串口连上esp32，进入REPL，在命令行创建一个无线AP看看是否能连上。
> * 蜂鸣器硬件看着跟我用的是一样的，供电是几伏？电源从哪里接的？
> * 热电偶我感觉多数是硬件本身问题了，你尝试换个模块或者探头看看是否有区别。

感谢您的指导。蜂鸣器有哒哒声但不影响使用，暂时先不管他了。热电偶用图上这种可以吗？这种探头是不是会影响敏感度呢？
[image]

dukeduck1984, 2021-02-06:
这个探头不合适，外面金属螺丝的热惯量太大了，不利于探头对于外界温度变化的快速反应，只适合于保持恒温的场景，比如冰箱的温控。最好使用玻璃封装的探头，然后用金手指胶带把探头贴在一块pcb上，参见readme里我烤箱的内部照片。

nmcq, 2021-02-06:
> 这个探头不合适，外面金属螺丝的热惯量太大了，不利于探头对于外界温度变化的快速反应，只适合于保持恒温的场景，比如冰箱的温控。最好使用玻璃封装的探头，然后用金手指胶带把探头贴在一块pcb上，参见readme里我烤箱的内部照片。

感谢您的答复。我也感觉探头的金属外壳有问题，影响反应速度。我重新 刷了固件，重新上传了程序 目前wifi已经可以用。

nmcq, 2021-02-06:
[image]
这样的探头可以吗。您的图片看不清样子

dukeduck1984, 2021-02-06:
探头可以，记得问一下导线是否耐高温

nmcq, 2021-02-06:
> 探头可以，记得问一下导线是否耐高温

谢谢。31855要比6675好一些吗？我准备再买一个控制板加探头。

dukeduck1984, 2021-02-06:
凡事问谷歌 :-) [Max6675 versus Max31855](https://forum.arduino.cc/index.php?topic=505463.0)

nmcq, 2021-02-06:
> 凡事问谷歌 :-) [Max6675 versus Max31855](https://forum.arduino.cc/index.php?topic=505463.0)
果然谷歌还是厉害的。百度不太行。谢谢啦

closed 2021-02-06 by nmcq

pastaclub, 2021-02-06:
I would say that it depends on your heater whether the probe with the screw is a good idea. If you work with a toaster, then you have to measure the air temperature and as Kaiyuan said, it is better to use a probe with less thermal inertia. However, if you are using a hot plate, you should directly measure the temperature of the plate, since it will have direct contact with the PCB. For this, the probe with screw will work fine.

That said, the screw is just an enclosure. If you already have this probe, you can carefully disassemble it and inside you will find the same thing as your taobao photo.

nmcq, 2021-02-06:
> 我会说，这取决于你的加热器是否与螺丝探头是一个好主意。如果你使用烤面包机，那么你必须测量空气温度，正如开元说的，最好使用热惯性较少的探头。但是，如果您使用的是热板，应直接测量板的温度，因为它会与 PCB 直接接触。为此，带螺钉的探头将正常工作。
> 
> 这就是说，螺丝只是一个外壳。如果你已经有这个探头，你可以仔细拆解它，里面你会发现同样的事情，你的淘宝照片。

thanks.Let  me have a try.