dukeduck1984/uReflowOven-Esp32-Micropython #16: 麻烦帮忙看看我这个曲线能实现焊接吗

nmcq, 2021-02-15:
[image]

> dukeduck1984，新年好，请教一下我的曲线可否焊接？

nmcq, 2021-02-15:
[image]
这是第二次的曲线，第一张那个预热阶段被挡住了。预热那儿曲线偏离较大。

dukeduck1984, 2021-02-15:
新年好！可以焊接的，另外你可以尝试把pid参数ki再稍稍调大一点点。

一开始预热不够快的问题我做了个解决方案在`beta2`里，有兴趣可测试下

dukeduck1984, 2021-02-15:
你可以去TB上买些陶瓷隔热棉，把烤箱的加热腔包裹起来，这样可以减小热惯量，一开始的加热会更迅速。但要操作时要注意安全，需要戴橡胶手套和护目镜，不要揉鼻子揉嘴巴，避免陶瓷棉的碎屑或者灰尘进到眼睛鼻子嘴巴里。我是用把大剪刀（张小泉那种铁剪刀）事先把陶瓷棉裁剪好，在烤箱加热腔外打上一些耐高温硅胶（TB上也有卖），接着把剪裁好的隔热棉贴上去，最后用宽的金手指胶带在陶瓷棉外包裹下，待硅胶凝固就固定好了。这样操作还有一个好处就是对电子器件比较好，不必承受烤箱加热带来的高温。

nmcq, 2021-02-15:
> 新年好！可以焊接的，另外你可以尝试把pid参数ki再稍稍调大一点点。
> 
> 一开始预热不够快的问题我做了个解决方案在`beta2`里，有兴趣可测试下

beta2在哪儿下载？

包隔热棉以前做烘焙时候想尝试来着。回头可以试一下。

dukeduck1984, 2021-02-15:
branch 'beta2'

nmcq, 2021-02-22:
找了个小板测试了一下，非常好用

dukeduck1984, 2021-02-22:
你试过branch beta2的版本吗？一开始升温不够快的问题是否改善？

closed 2021-04-29 by dukeduck1984