dukeduck1984/uReflowOven-Esp32-Micropython #20: Status of the project today and doubts.

serlancelot, 2022-09-30:
First of all thanks for sharing all this (there aren't many people like that, or at least I don't know them)
I have been searching a lot on the internet and there are very cool projects but with screens of 60 euros (which used to be 25...) and others with more normal screens type controleo3 which I decided to do and after buying everything on aliexpress the samd21j18 is missing from the stores... and now I have material which I don't want to waste.... and there is good in these questions.
The project is still alive? if so, is it possible to use a 4" screen with the ILI9488 and use a max31865?
I have bought the esp (https://es.aliexpress.com/item/32855652152.html?gatewayAdapt=glo2esp) that indicates to carry out the project.
With this python thing I get lost... what versions should I have installed for it to work?
Thank you so much for everything!! All the best.

pastaclub, 2022-10-01:
The project is alive, but there isn't really any news since the contributors have their ovens and hotplates which are working fine and there is no need to change anything. Certainly you can use different temperature sensors and different screens, but you will have to change the code. If your temperature sensor is similar, it would probably be an easy task. The screen will be more work I guess, because yours has a higher resolution. You will need more RAM for the frame buffer and the GUI may look broken and need fixing too.

closed 2022-10-01 by pastaclub

dukeduck1984, 2022-10-01:
@serlancelot Pls use the firmware included in the `FIRMWARE` folder.

serlancelot, 2022-10-01:
Hello, thanks for the answers, install the bin file from the firmware folder when I get the esp I will try it to see if I can install it I will follow the instructions and see what I do if I don't mess it up